radioconda/radioconda-installer #87: 2024.01.26 Release Not Installing on Intel-based Mac

Stroke44, 2024-02-16:
I'm trying to install radioconda from an existing conda prompt on an Intel-based Mac.  It has worked fine with many previous versions, but it seems that the packages cannot be found:

`(base) john@Magni ~ % conda create -n radioconda -c conda-forge -c ryanvolz --only-deps radioconda
Collecting package metadata (current_repodata.json): done
Solving environment: unsuccessful attempt using repodata from current_repodata.json, retrying with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: - `

ryanvolz, 2024-02-16:
Admittedly I don't have a Mac to test this on, but I think I can simulate the environment solution with the `CONDA_SUBDIR` environment variable:
```
CONDA_SUBDIR=osx-64 conda create -n test -c conda-forge -c ryanvolz --only-deps radioconda
```
That works for me, either using conda 23.10.0 or mamba 1.5.3 in my base environment.

So this sounds like either a conda issue/bug where perhaps you need to upgrade (or even downgrade?). I *think* conda has moved away from using `current_repodata.json` for package metadata and my text output does not mention it, so probably an upgrade is needed. I'm not on the latest version though, so maybe a downgrade helps.

If having conda solve for the environment remains an issue, you should be able to bypass the solver and create the environment from a static package list using
```
conda create -n radioconda --file https://github.com/ryanvolz/radioconda/releases/latest/download/radioconda-osx-64.lock
```

Stroke44, 2024-02-17:
I made sure my Anaconda distribution was up to date.  I still had the same problem.  I tried the following:
`conda create -n radioconda --file https://github.com/ryanvolz/radioconda/releases/latest/download/radioconda-osx-64.lock`
and it worked!  Everything installed fine.  Thank you for the suggestion!

Here's my conda information for reference:
     active environment : base
    active env location : /Users/john/anaconda3
            shell level : 1
       user config file : /Users/john/.condarc
 populated config files : /Users/john/.condarc
          conda version : 23.7.4
    conda-build version : 3.26.1
         python version : 3.11.5.final.0
       virtual packages : __archspec=1=x86_64
                          __osx=10.16=0
                          __unix=0=0
       base environment : /Users/john/anaconda3  (writable)
      conda av data dir : /Users/john/anaconda3/etc/conda
  conda av metadata url : None
           channel URLs : https://repo.anaconda.com/pkgs/main/osx-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/osx-64
                          https://repo.anaconda.com/pkgs/r/noarch
          package cache : /Users/john/anaconda3/pkgs
                          /Users/john/.conda/pkgs
       envs directories : /Users/john/anaconda3/envs
                          /Users/john/.conda/envs
               platform : osx-64
             user-agent : conda/23.7.4 requests/2.31.0 CPython/3.11.5 Darwin/20.6.0 OSX/10.16 aau/0.4.2 c/p6FnaI080BGtKpspaRLnEg s/ujswx0CErQRuqtiCHarmkg e/lqWq1fvM9SmBf4kIgPetSw
                UID:GID : 501:20
             netrc file : None
           offline mode : False

closed 2024-02-17 by Stroke44