radioconda/radioconda-installer #90: Source code of the Signal Generatoe FMCW Out of the tree block.

patilraj1707, 2024-03-22:
I was trying to implement a FMCW radar in gnu radio using pluto SDR by making my own block but then I came to know that there exists an block in windows that gives you the chirp signal, so it would be very helpfull if you would be able to provide me the yaml and the python files or atlest tell me the location of the files in my installation.

ryanvolz, 2024-03-22:
That block is part of the gr-radar OOT module, and the source code is on GitHub at https://github.com/kit-cel/gr-radar/blob/master/lib/signal_generator_fmcw_c_impl.cc. It's not a Python block, but C++. The GRC YAML file is here https://github.com/kit-cel/gr-radar/blob/master/grc/radar_signal_generator_fmcw_c.block.yml.

closed 2024-03-22 by ryanvolz

patilraj1707, 2024-03-22:
Can you share a example imolementation of this block because some parameters of the block are confusing to me.

ryanvolz, 2024-03-22:
Sorry, I didn't write the block not have I ever used it, so I don't know anything more than what I've already linked. Looking at the code, it appears that the block outputs 3 blocks of samples of specified lengths before repeating:

1. `Samples CW` samples of CW at the specified `Frequency CW`
2. `Samples up-chirp` samples of of the signal linearly increasing in frequency starting at `Frequency CW`  and ending at `Frequency CW` + `Sweep frequency`
3. `Samples down-chirp` samples of of the signal linearly decreasing in frequency starting at `Frequency CW` + `Sweep frequency` and ending at `Frequency CW`

patilraj1707, 2024-03-23:
okay thank you very much