radioconda/radioconda-installer #92: Typelib file for namespace 'GObject', version '2.0' not found

vaughanatworld, 2024-04-20:
radioconda installed on Debian 11
Error I'm getting when starting gnuradio-companion:
ImportError

Failed to initialize GTK. If you are running over ssh, did you enable X forwarding and start ssh with -X?

(Typelib file for namespace 'GObject', version '2.0' not found)

It seems there is a missing typelib file for GTK, libgobject-2.0.so
Googling around, I see this problem occurring for other applications but mainly for radioconda on windows and linux.
Has anyone found a workaround for this issue?
Thanks

ryanvolz, 2024-04-26:
Are you using the latest release without upgrading any packages or have you upgraded packages since installing? If the latter, there have been some problems with the latest `glib` package on conda-forge, so it is possible you are running into those issues. As far as I know, the `glib` problems should be fixed now if you have the latest packages as of a day or two ago. If you're just using the latest radioconda release without any changes, then that's something else and it would be good to know.

vaughanatworld, 2024-04-29:
I'll take a look at the latest glib package. 
I did the install a couple of weeks ago and upgraded the whole machine and verified I had the latest glib.
I haven't messed with it since. I'll post back here when I've re-upgrade my glib. 
Debian packages can be a little slow to upgrade so I'll double check the GTK website too.
Thanks for the response.
Brian