radioconda/radioconda-installer #97: MacOS version support

hjimenez-ni, 2024-06-07:
I tried installing the latest release [2024.05.29](https://github.com/ryanvolz/radioconda/releases/tag/2024.05.29) on my old MBP 15" Mid-2015 (Intel-based). The latest supported OS for this Mac is Monterey 12.7.5.
[image]

I got an error because the .pkg is not supported in my OS. I also tried installing an older release (early 2021) without luck.
Where can I find a MacOS compatibility chart for the releases? (I was not able to find one in my search through the repo).

Thanks a lot for the support and great contribution to the SDR community.

ryanvolz, 2024-06-10:
Radioconda's macOS support necessarily follows that of conda-forge, from which the packages are built. Conda-forge strives for wide compatibility, and should currently be compatible with macOS versions back to 10.13. Older releases I think had support back to 10.9. Regardless, they all should work fine with anything newer like 12.7.5.

You said you have an Intel Mac. Can you confirm that you've tired the [x86_64 version](https://github.com/ryanvolz/radioconda/releases/download/2024.05.29/radioconda-2024.05.29-MacOSX-x86_64.pkg) of the package? Do you have better luck if you use the [command line installer](https://github.com/ryanvolz/radioconda/releases/download/2024.05.29/radioconda-2024.05.29-MacOSX-x86_64.sh)? I don't have a Mac and so have never tested any of this personally on those systems, but this is the first report I've heard about something not working so I'd definitely like to figure out what's going on.

The "caution" banner at the top of this page, which should also apply to radioconda, may also be relevant for you: https://docs.conda.io/projects/conda/en/latest/user-guide/install/macos.html

serhiiyanholairwayz, 2024-08-04:
sorry to bother you with a perhaps not the most intelligent question, but I'm new to using macOS. radioconda installed without errors (m2 max sonoma 14.6). however hackrf is not detected at all. I have two of them, none of them is detected. at the same time, everything is fine with them on a laptop with Linux, they work.
from the steps described in the installation, I saw the following:

- sudo ln -s $CONDA_PREFIX/lib/udev/rules.d/53-hackrf.rules /etc/udev/rules.d/53-radioconda-hackrf.rules
- sudo udevadm control --reload
- sudo udevadm trigger

but as far as I understand it only applies to linux. there is no need to do it on the mac, it should be apply automatically..
I just don't know what to do anymore... pls, help. thank you

littleyeson, 2025-04-28:
> 我尝试在我的旧款 MBP 15 英寸（2015 年中）上安装最新版本[2024.05.29 。这台 Mac 最新支持的操作系统是 Monterey 12.7.5。](https://github.com/ryanvolz/radioconda/releases/tag/2024.05.29) <img alt="屏幕截图 2024-06-07 下午 12:37:09" width="246" src="https://private-user-images.githubusercontent.com/31545256/337742945-91bbd373-7157-4e83-83ed-75ab8ccd439d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDU4MDcxNTMsIm5iZiI6MTc0NTgwNjg1MywicGF0aCI6Ii8zMTU0NTI1Ni8zMzc3NDI5NDUtOTFiYmQzNzMtNzE1Ny00ZTgzLTgzZWQtNzVhYjhjY2Q0MzlkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA0MjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNDI4VDAyMjA1M1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQ5NGU3Yzc1MThlYjE3MjEyNTNjYmU3ZjI0OWI0OGQ0NjU2YTViM2IyNmFkMDYyYzNlYWIyYWI0MjBlZDBlNDcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.oltcLuq6D7mu3n-wfdU8lrA72IHEkfMcBPqNNr8AyQU">
> 
> 我遇到了一个错误，因为我的操作系统不支持 .pkg 文件。我还尝试安装一个旧版本（2021 年初），但没有成功。 我在哪里可以找到这些版本的 MacOS 兼容性图表？（我在仓库中搜索后没有找到）。
> 
> 非常感谢您对 SDR 社区的支持和巨大贡献。

The same problem.How to fix it?