radioconda/radioconda-installer #98: "The installation failed" while installing radioconda 2024.05.29 macOS package

deviantintegral, 2024-06-13:
I'm attempting to install radioconda on macOS 14.5 with the arm64 packages. After extracting everything and running postinstall scripts, the installer shows this error:

[image]

As the logs below say, I don't have a `.conda` directory.

Full logs: https://gist.github.com/deviantintegral/055cee8a01a82e8943d70deea3019a31

```
Jun 12 21:32:14 acutus package_script_service[95219]: ./postinstall: Linking gnuradio-adsb-0.0.0.20240508.dev+gafd7fabda-unix_pyha6c0e43_0
Jun 12 21:32:14 acutus package_script_service[95219]: ./postinstall: Linking gqrx-2.17.5-py310h1a81a2b_4
Jun 12 21:32:14 acutus package_script_service[95219]: ./postinstall: Linking conda-libmamba-solver-24.1.0-pyhd8ed1ab_0
Jun 12 21:32:14 acutus package_script_service[95219]: ./postinstall: Linking conda-24.5.0-py310hbe9552e_0
Jun 12 21:32:14 acutus package_script_service[95219]: ./postinstall: Linking mamba-1.5.8-py310ha5d4528_0
Jun 12 21:32:14 acutus package_script_service[95219]: ./postinstall: Transaction finished
Jun 12 21:32:15 acutus package_script_service[95219]: ./postinstall: Python 3.10.14
Jun 12 21:32:19 acutus package_script_service[95219]: ./postinstall: chown: /Users/andrew/.conda: No such file or directory
Jun 12 21:32:19 acutus package_script_service[95219]: Responsibility set back to self.
Jun 12 21:32:19 acutus installd[95218]: PackageKit: releasing backupd
Jun 12 21:32:19 acutus installd[95218]: PackageKit: allow user idle system sleep
Jun 12 21:32:19 acutus installd[95218]: PackageKit: Install Failed: Error Domain=PKInstallErrorDomain Code=112 "An error occurred while running scripts from the package “radioconda-2024.05.29-MacOSX-arm64.pkg”." UserInfo={NSFilePath=./postinstall, NSURL=file:///Users/andrew/Downloads/radioconda-2024.05.29-MacOSX-arm64.pkg#run_installation.pkg, PKInstallPackageIdentifier=io.continuum.pkg.run_installation, NSLocalizedDescription=An error occurred while running scripts from the package “radioconda-2024.05.29-MacOSX-arm64.pkg”.} {
	    NSFilePath = "./postinstall";
	    NSLocalizedDescription = "An error occurred while running scripts from the package \U201cradioconda-2024.05.29-MacOSX-arm64.pkg\U201d.";
	    NSURL = "file:///Users/andrew/Downloads/radioconda-2024.05.29-MacOSX-arm64.pkg#run_installation.pkg";
	    PKInstallPackageIdentifier = "io.continuum.pkg.run_installation";
	}
Jun 12 21:32:19 acutus installd[95218]: PackageKit: Cleared responsibility for install from 96472.
Jun 12 21:32:19 acutus installd[95218]: PackageKit: Cleared permissions on Installer.app
Jun 12 21:32:19 acutus Installer[96472]: install:didFailWithError:Error Domain=PKInstallErrorDomain Code=112 "An error occurred while running scripts from the package “radioconda-2024.05.29-MacOSX-arm64.pkg”." UserInfo={NSFilePath=./postinstall, NSURL=file:///Users/andrew/Downloads/radioconda-2024.05.29-MacOSX-arm64.pkg#run_installation.pkg, PKInstallPackageIdentifier=io.continuum.pkg.run_installation, NSLocalizedDescription=An error occurred while running scripts from the package “radioconda-2024.05.29-MacOSX-arm64.pkg”.}
Jun 12 21:32:19 acutus installd[95218]: PackageKit: Removing client PKInstallDaemonClient pid=96472, uid=501 (/System/Library/CoreServices/Installer.app/Contents/MacOS/Installer)
Jun 12 21:32:20 acutus Installer[96472]: PackageKit: Bypassing the enforcement checker since the destination is read/write
Jun 12 21:32:20 acutus Installer[96472]: Install failed: The Installer encountered an error that caused the installation to fail. Contact the software manufacturer for assistance.
Jun 12 21:32:20 acutus Installer[96472]: IFDInstallController F166A0 state = 8
Jun 12 21:32:20 acutus Installer[96472]: Displaying 'Install Failed' UI.
Jun 12 21:32:20 acutus Installer[96472]: 'Install Failed' UI displayed message:'The Installer encountered an error that caused the installation to fail. Contact the software manufacturer for assistance.'.
```

ryanvolz, 2024-06-13:
Interesting. Aside from the message and no `.conda` directory, does the installation still work?

This seems to be related to a [recent change in conda constructor](https://github.com/conda/constructor/pull/784). A question relevant to that change: did you install to the default location, or did you change it to something else?

deviantintegral, 2024-06-15:
I ended up trying the bash installer, which had no errors. I would have thought the code would be the same at this part of the installation.

It looks like this errors out before it sets up the shell, so the install doesn't work as far as most users would be concerned.

JimMerkel, 2024-07-03:
I had a similar situation with macOS 14.5 on x86_64 (Intel).  Graphical Installer failed, but Command-line installer worked fine.

jaimergp, 2024-07-12:
Added https://github.com/conda/constructor/pull/820 to fix the regression. For now avoid building an installer with constructor 3.8.0.

thanosdr, 2024-12-12:
Same error here. maybe its obvious to some but for those who are computer savvy yet not programmers;

Fix was:
**in terminal** 
install anaconda:                          $brew install anaconda
remove old instal directory:          $rm -rf ~/radioconda
change permissions on .conda     $sudo chown -R "$USER" "${HOME}/.conda"
**in finder**
run installer again

ryanvolz, 2025-03-14:
Should be fixed now with the latest release. Finally!

closed 2025-03-14 by ryanvolz