radioconda/radioconda-installer #99: Dynamic Channel model block issue

MaxCrt, 2024-06-16:
After using GNUradio 3.7 and 3.8, I installed the latest Gnuradio version 3.10.10.

I tried several times to utilize the Dynamic Channel model block on 3.10.10, without success.
While on 3.7 and 3.8 it works always.

Here you find the error reported:
  File "C:\GNU Radio 3.10 my files\DmrIQPlayer.py", line 213, in __init__
    self.channels_dynamic_channel_model_0 = channels.dynamic_channel_model(
                                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: __init__(): incompatible constructor arguments. The following argument types are supported:
    1. gnuradio.channels.channels_python.dynamic_channel_model(samp_rate: float, sro_std_dev: float, sro_max_dev: float, cfo_std_dev: float, cfo_max_dev: float, N: int, doppler_freq: float, LOS_model: bool, K: float, delays: list[float], mags: list[float], ntaps_mpath: int, noise_amp: float, noise_seed: float)

Invoked with: 93750, 1, 10, 1, 10, 6, 10, False, 4.0, 0.0, 5.0, 20.0, 50.0, 80.0, 1.0, 0.8, 0.7, 0.6, 0.1, 8, 0, 0

>>> Done (return code 1)

Could you check that block works properly on 3.10.10 ?

Thanks in advance

BR,
Max IK1RGI

ryanvolz, 2024-06-18:
I see this as well. It looks like a proper GNU Radio bug. Can you post this as an issue upstream at https://github.com/gnuradio/gnuradio/issues?

MaxCrt, 2024-06-19:
Hi Ryan,  gnuradio thread found the issue in the code. Compared to the old same lock, they forgot to put the list of array's value you type between []. The code is not adding the "[....]".