radioconda/radioconda-installer #103: HackRF not detected using macOS

ryanvolz, 2024-08-05:
sorry to bother you with a perhaps not the most intelligent question, but I'm new to using macOS. radioconda installed without errors (m2 max sonoma 14.6). however hackrf is not detected at all. I have two of them, none of them is detected. at the same time, everything is fine with them on a laptop with Linux, they work.
from the steps described in the installation, I saw the following:

- sudo ln -s $CONDA_PREFIX/lib/udev/rules.d/53-hackrf.rules /etc/udev/rules.d/53-radioconda-hackrf.rules
- sudo udevadm control --reload
- sudo udevadm trigger

but as far as I understand it only applies to linux. there is no need to do it on the mac, it should be apply automatically..
I just don't know what to do anymore... pls, help. thank you

_Originally posted by @serhiiyanholairwayz in https://github.com/ryanvolz/radioconda/issues/97#issuecomment-2267640574_

serhiiyanholairwayz, 2024-08-05:
@ryanvolz, thank you for quick reply.
hackrf_info show that 
hackrf_info version: 2024.02.1
libhackrf version: 2024.02.1 (0.9)
No HackRF boards found.

also, sometimes I can see the next output using
sudo dmesg | grep hackrf
[ 3619.032024]: Waking up reference: 153Thread waiting on reference 153 woke upevaluation result: 153, library, allowed, cache, 1722689181, 0, 90201ae9ab0f59a3, /opt/homebrew/Cellar/libusb/HEAD-467b6a8/lib/libusb-1.0.0.dylibhackrf_info@: AppleUSBHostUserClient::openGated: failed to open USB3.0 Hub @02200000: provider is already opened for exclusive access by AppleUSB30Hub

ryanvolz, 2024-08-05:
It seems like some sort of hardware permission error. I don't consider it a good sign that the dmesg output references `/opt/homebrew/Cellar/libusb/HEAD-467b6a8/lib/libusb-1.0.0.dylib`, which would imply that it is using a libusb installed via homebrew and not the one install via conda. Maybe that mismatch is causing an issue? Can you uninstall the homebrew libusb, or would that break something else?

Alternatively, you could try to see if using the hackrf tools [installed via homebrew](https://hackrf.readthedocs.io/en/latest/installing_hackrf_software.html#os-x-10-5-homebrew) would allow access, which would narrow this down as a conda-specific problem and not a general hardware/OS problem.

smishr, 2025-03-22:
Hi @serhiiyanholairwayz I had same issue when setting up HackRF on MacOS.

For me the issue was resolved by using a different USB cable used to connect the laptop to HackRF. See [here](https://www.reddit.com/r/hackrf/comments/1fek5bs/hackrf_mode_on_portapack_not_working/)