radioconda/radioconda-installer #110: portaudio should be updated to 19.7.0

kgarrels, 2024-10-27:
Hi,

portaudio 19.6.0 causes problems on macos' newer versions.

Could we upgrade to 19.7.0?
https://github.com/PortAudio/portaudio/releases

Thanks for the great work!
kai

ryanvolz, 2024-10-29:
Good to know, thanks! The `portaudio` package is coming from https://github.com/conda-forge/portaudio-feedstock, which indeed has not been updated to 19.7.0 (even though that's been release for a long while). I'm not a maintainer there, but I'll poke them to try to get it updated.

kgarrels, 2024-11-03:
it seems 19.7.0. is available, can I somehow work around this error?

```
mamba install portaudio=19.7.0 --channel conda-forge

Looking for: ['portaudio=19.7.0']

warning  libmamba Cache file "/Users/kai/radioconda/pkgs/cache/a850f475.json" was modified by another program
warning  libmamba Cache file "/Users/kai/radioconda/pkgs/cache/09cdf8bf.json" was modified by another program
warning  libmamba Cache file "/Users/kai/radioconda/pkgs/cache/fa49af39.json" was modified by another program
ryanvolz/osx-arm64 (check zst)                      Checked  1.6s
warning  libmamba Cache file "/Users/kai/radioconda/pkgs/cache/d8f1edbb.json" was modified by another program
ryanvolz/noarch (check zst)                         Checked  0.4s
ryanvolz/noarch                                      3.5kB @  14.5kB/s  0.2s
ryanvolz/osx-arm64                                  38.0kB @ 102.0kB/s  0.4s
conda-forge/noarch                                  17.1MB @   7.3MB/s  2.4s
conda-forge/osx-arm64                               12.7MB @   5.1MB/s  2.5s

Pinned packages:
  - python 3.10.*

warning  libmamba Added empty dependency for problem type SOLVER_RULE_UPDATE
Could not solve for environment specs
The following packages are incompatible
├─ gnuradio-core is installable with the potential options
│  ├─ gnuradio-core [3.10.0.0|3.10.1.0|...|3.9.8.0] would require
│  │  └─ portaudio >=19.6.0,<19.7.0a0 , which can be installed;
│  ├─ gnuradio-core [3.10.0.0|3.10.1.0|...|3.9.8.0] would require
│  │  └─ python >=3.8,<3.9.0a0 , which can be installed;
│  ├─ gnuradio-core [3.10.0.0|3.10.1.0|...|3.9.8.0] would require
│  │  └─ python >=3.9,<3.10.0a0 , which can be installed;
│  ├─ gnuradio-core [3.10.10.0|3.10.11.0|...|3.9.8.0] would require
│  │  └─ python >=3.11,<3.12.0a0 , which can be installed;
│  └─ gnuradio-core [3.10.10.0|3.10.11.0|...|3.10.9.2] would require
│     └─ python >=3.12,<3.13.0a0 , which can be installed;
└─ portaudio 19.7.0**  is not installable because it conflicts with any installable versions previously reported.
```

closed 2025-03-01 by kgarrels