radioconda/radioconda-installer #111: RTL-SDR instructions too brief

micsthepick, 2024-10-29:
The RTL-SDR blog page includes the manufacturer and device id for the SDR, and for my case I needed to verify that, as 
[image]
in this image, my rtl-sdr was the second entry.

ryanvolz, 2024-10-29:
How about changing the text from

> [Install the WinUSB driver with Zadig](https://github.com/ryanvolz/radioconda?tab=readme-ov-file#installing-the-winusb-driver-with-zadig), selecting the device that is called "Bulk-In, Interface (Interface 0)".

to

> [Install the WinUSB driver with Zadig](https://github.com/ryanvolz/radioconda?tab=readme-ov-file#installing-the-winusb-driver-with-zadig), selecting the device that is called "Bulk-In, Interface" and verifying the manufacturer and device ID if there are multiple options.

?

micsthepick, 2024-10-29:
Never mind, it was seemingly working but I needed to actually follow
the instructions verbatim, and thus had to do the signalseverywhere USB
error -12 fix

You may close this issue

closed 2024-10-30 by micsthepick