radioconda/radioconda-installer #112: Windows & RTLSDR - RuntimeError: SoapySDR::Device::make() no match

OE3JGW, 2024-10-29:
Hello, 
newcomer here. awesome project i love it!
But i cant even begin learning all that stuff, cause i get the following error..
Im on Win11 x64 and installed radioconda with the exe file. 

Can someone help me please :/ 

**Error**
```
Executing: d:\python_projects\pySDRscan\venv\python.exe -u C:\Users\waiss\FM_sdr.py

Traceback (most recent call last):
  File "C:\Users\waiss\FM_sdr.py", line 216, in <module>
    main()
  File "C:\Users\waiss\FM_sdr.py", line 193, in main
    tb = top_block_cls()
  File "C:\Users\waiss\FM_sdr.py", line 95, in __init__
    self.soapy_rtlsdr_source_0 = soapy.source(dev, "fc32", 1, 'driver=rtlsdr,serial=77771111153705700',
RuntimeError: SoapySDR::Device::make() no match

>>> Done (return code 1)
```

**SoapySDRUtil**
```
(base) PS C:\Users\waiss> SoapySDRUtil --probe="driver=rtlsdr"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device driver=rtlsdr
[WARNING] SoapyVOLKConverters: no VOLK config file found. Run volk_profile for best performance.
Found Rafael Micro R820T tuner
[INFO] Opening Generic RTL2832U :: 77771111153705700...
Found Rafael Micro R820T tuner

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=RTLSDR
  hardware=R820T
  index=0
  origin=https://github.com/pothosware/SoapyRTLSDR

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 0 Tx
  Timestamps: YES
  Time sources: sw_ticks
  Other Settings:
     * Direct Sampling - RTL-SDR Direct Sampling Mode
       [key=direct_samp, default=0, type=string, options=(0, 1, 2)]
     * Offset Tune - RTL-SDR Offset Tuning Mode
       [key=offset_tune, default=false, type=bool]
     * I/Q Swap - RTL-SDR I/Q Swap Mode
       [key=iq_swap, default=false, type=bool]
     * Digital AGC - RTL-SDR digital AGC Mode
       [key=digital_agc, default=false, type=bool]
     * Bias Tee - RTL-SDR Blog V.3 Bias-Tee Mode
       [key=biastee, default=false, type=bool]

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: NO
  Supports AGC: YES
  Stream formats: CS8, CS16, CF32
  Native format: CS8 [full-scale=128]
  Stream args:
     * Buffer Size - Number of bytes per buffer, multiples of 512 only.
       [key=bufflen, units=bytes, default=262144, type=int]
     * Ring buffers - Number of buffers in the ring.
       [key=buffers, units=buffers, default=15, type=int]
     * Async buffers - Number of async usb buffers (advanced).
       [key=asyncBuffs, units=buffers, default=0, type=int]
  Antennas: RX
  Full gain range: [0, 49.6] dB
    TUNER gain range: [0, 49.6] dB
  Full freq range: [23.999, 1764] MHz
    RF freq range: [24, 1764] MHz
    CORR freq range: [-0.001, 0.001] MHz
  Sample rates: [0.225001, 0.3], [0.900001, 3.2] MSps
  Filter bandwidths: [0, 8] MHz
```
 
### Screenshot
[image]

OE3JGW, 2024-10-29:
deinstalled all pothos, python versions, radioconda (i had an old version installed aswell it looks like) and removed everything i was able to find about relatable stuff. 

restarted, reinstalled radioconda - now its working. 
i guess soapy was somehow installed twice, i just was not able to find it.

ryanvolz, 2024-10-29:
Looks like `FM_sdr.py` script was running in the `pySDRscan` venv with the Python that comes there instead of the radioconda Python, which I guess brings a different Soapy installation that was interfering. Uninstalling clearly helped, but I think it could have worked if you'd run that Python script from within the radioconda prompt (and its activated conda environment).

OE3JGW, 2024-10-30:
Thanks for your reply ryanvoiz.
Thats true and i came to the same conclusion at the end,  but for whatever reason, this was not enough. 
i still really dont know what exactly caused the issue, but it works now which is the important thing :) 
already got a runny wbfm receiver, its so much fun to tinker with it :D