radioconda/radioconda-installer #122: Virus found installing radioconda

bgerzoff, 2025-03-26:
Bitdefender reported the following:

Feature:
Antivirus

The file C:\ProgramData\radioconda\pkgs\libairspyhf-1.6.8-h2466b09_1\Library\bin\airspyhf.dll is infected with Gen:Variant.Lazy.189143 and was moved to quarantine. It is recommended that you run a System Scan to make sure your system is clean.

ryanvolz, 2025-03-26:
This seems likely to be a false positive. That DLL is built from [a public feedstock](https://github.com/conda-forge/airspyhf-feedstock), and one can view the build logs [here](https://dev.azure.com/conda-forge/feedstock-builds/_build/results?buildId=1195143&view=logs&jobId=fe7faf59-69bb-5487-b8fd-a35a7bda3a51&j=fe7faf59-69bb-5487-b8fd-a35a7bda3a51&t=55731638-dd49-5144-2129-5101b9dd1960). Unless the anaconda.org server or the Github installer artifact was compromised and the DLL replaced with a malicious one (which I consider unlikely, but you might assess the risks differently), then what that build produced is what you have.

If you agree with that assessment, then this looks like a next step: https://www.bitdefender.com/business/support/en/77209-343054-resolving-legitimate-applications-detected-as-threats-by-bitdefender.html.

bgerzoff, 2025-03-27:
Thanks for the quick reply.

Bob