radioconda/radioconda-installer #133: GR-DRM

dgregsontcl, 2025-08-29:
The DRM community is about to take off.  Several countries have recently adopted the format (Indonesia, Pakistan, Tanzania) India and China have been using it for several years.   Affordable Receivers are about to hit the market, and it would be very helpful to have GR-DRM included in Radioconda so community radio experimenters could test and develop.   This would be helpful in all environments, but most helpful in the Windows environment.   Ref:  https://github.com/kit-cel/gr-drm

ryanvolz, 2025-08-30:
Thanks for bumping the visibility of this. I can prioritize DRM support, but the kit-cel gr-drm looks like it hasn't been updated in a long time and doesn't support GNU Radio newer than 3.7. A forward port is not out of the question, but it's a bit more work and I'm wondering if there is any more recent software with DRM support that would be better to include. Are you aware of anything?

dgregsontcl, 2025-08-31:
Creating multiplex's would be the most helpful to learn and test.

Since gr-drm was released, the AAC low complexity license expired,  This makes the Multiplex using this or Opus viable and a great way to test.  Red Hat, google and others have included aac in their distros.

Ref: AI search,

Gemini

2.5 Pro

Conversation with Gemini

"The most widely used profile, AAC-LC (Low Complexity), is now largely considered free from patent restrictions in the United States. While there isn't a definitive universal date, a significant portion of the essential patents for AAC-LC expired by 2017. This assessment was notably put into practice by companies like Red Hat, which began including an AAC-LC encoder in its Linux distribution around that time after its legal team deemed the core patents to have expired."

AAC-LC is fine for testing and use with one channel drm30.  DRM plus is even better because of the available bandwidth.

Opus is always safe for testing, but only software decoders would work for decoding as no hardware vendors support opus.